Foxies-CSTL/Marlin_2.1.x #2: Add support for the original wifi firmware

Milkman337, 2022-07-22:
### Is your feature request related to a problem? Please describe.

I cant get the esp32 new wifi firmware to run

### Describe the feature you want

Please make marlin compatible with the qqs pro stock wifi firmware

Foxies-CSTL, 2022-07-23:
The firmware of the Wifi module is available on this page of my Wifi. 
https://github.com/Foxies-CSTL/Marlin_2.1.x/wiki/5.Firmware-Wifi
To talk with Marlin, you need to enable an additional communication channel (#define SERIAL_PORT_2 1) in the Marlin source or use a binary with the "W" option.

Milkman337, 2022-07-23:
but the wifi firmware from here(https://github.com/Foxies-CSTL/Marlin_2.1.x/wiki/5.Firmware-Wifi) doesnt work it says some error

Foxies-CSTL, 2022-08-29:
> also note that the qqs pro has an esp12s in it
Yes, I know that.

Now that version 2.1.1 is released, I will have some time to redo firmwares for the Wifi module with the last ESP3D source.

Foxies-CSTL, 2022-09-06:
The firmwares for the WIFI module will be updated. I have just finished different tests between remote hosts for monitoring and file transfer.
I used a TFT35v1 for the MKS(lvgl) UI and the COLOR UI from Marlin. I will make a table to show the capabilities.

Foxies-CSTL, 2022-09-08:
Update firmware ESP3Dv2.1/v3 for UI COLOR (Stock) and UI MKS (ex:TFT35v1)
- TFT_COLOR_UI (monitoring) 
- TFT_LVGL_UI(Monitoring+TransfertData)
Note: Need a firmware with option "w" and "I" instead "W".

Close this issue

closed 2022-09-08 by Foxies-CSTL