Foxies-CSTL/Marlin_2.1.x #3: Stuck on Booting screen [BUG]

D1siel, 2022-07-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When Q5 with robin nano 1.2 board get flashed with the Q5_8+9CWBmPULR-Robin_nano (LINK:https://github.com/Foxies-CSTL/Marlin_2.1.x/blob/Firmwares/Q5/Q5_8%2B9CWBmPULR-Robin_nano.bin) that got uploaded yesterday it boots normally the first time after flash. But fails to boot past the "Booting..." the second time you start the machine. I am using the 2.1.1 bug fix that should fix the SD card misbehaving. and then I would like the MPC that you added in ;).

I am running 3xTMC2208 and TMC2209 for E with a BMG. Everything besides that is stuck.

### Bug Timeline

Just updated 24-07-2022

### Expected behavior

I expected to get into the firmware or like the homescreen. Just anything beyond the green on black "booting...." sequence.

### Actual behavior

Stuck in booting logo. Never gets to the homescreen.

### Steps to Reproduce

1. Flash your Q5 with https://github.com/Foxies-CSTL/Marlin_2.1.x/blob/Firmwares/Q5/Q5_8%2B9CWBmPULR-Robin_nano.bin
2. Turn off printer and pull sdcard out
3. Turn on printer
4. Welcome to the world of booting :D

### Version of Marlin Firmware

2.1.1.

### Printer model

Flsun Q5

### Electronics

Stock old board robin nano 1.2

### Add-ons

TMC2209 for E0 and a BMG extruder

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ulipo, 2022-07-30:
same behaviour here, ecxept I'm using Q5_8+SCWTmPULR-Robin_nano.bin
Boots and works normally with Q5_8+SCWTPULR-Robin_nano.bin

Foxies-CSTL, 2022-08-29:
Hi @D1siel @ulipo,
Version 2.1 and even the BugFix was really unstable for different kind of printers. The release of 2.1.1 should solve your problem. For the story about the screen blocking with the MoBo Hisspeedv1 (GD32), you have to remove the shielding from the screen ribbon. I have added a note in the ReadMe of the QQSP firmwares.

ulipo, 2022-08-29:
Seems to works so far, I'm printing a calibration cube right now.