Foxies-CSTL/Marlin_2.1.x #4: MKS SLOT2 SD [BUG]

riccalisci, 2022-08-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

``the SD card inserted in the external SD reader is recognized correctly but when printing is started, the printer resets when the set temperature is reached.
[https://youtu.be/gPnYbzBWXPY](url)

### Bug Timeline

problem also found in 2.1.0 version 

### Version of Marlin Firmware

2.1.1

### Printer model

Flsun Q5

### Electronics

MKS Robin Nano 1.3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

IdeaMaker

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[confguration.zip](https://github.com/Foxies-CSTL/Marlin_2.1.x/files/9269063/confguration.zip)

Foxies-CSTL, 2022-08-29:
Hi @riccalisci  
Sorry but I was busy with the preparation of the 2.1.1. Does the problem persist with this version?

riccalisci, 2022-08-29:
yes, i'have tested also 2.1.1 (beta version) and problem still exist

Foxies-CSTL, 2022-08-29:
> yes, i'have tested also 2.1.1 (beta version) and problem still exist

I'm talking about the 2.1.1 version of 27/08/2021, not the bugfix. 
Did you perform a Nozzle PID in the preparation of your printer after leveling for PLA?

riccalisci, 2022-08-29:
i can try a new stabile versione 2.1.1, could it have been fixed?
Yes, i've perform a nozzle and bed PID.

riccalisci, 2022-09-16:
i've tried with 2.1.1 and 2.0.9.3 versione, same issue.

Foxies-CSTL, 2022-10-31:
Hi,
As you are using a Q5 with MoBo nanoV1.3, I think that in the definition of the pins, the part where the use of external extension for a card reader (card socket) is missing. When I made my pin definition, I added this part because the HiSpeedV1 has this connector but I did not validate it with hardware.

MoBo NanoV1.x
[image]

MoBo HiSpeedV1
[image]

Maybe adding this part in the pin file, could help you.

riccalisci, 2022-11-07:
Thanks!
I can try it,  but i saw that the pins are the same from schematics.
[image]
[MKS Robin Nano V1.3_002 SCH.pdf](https://github.com/Foxies-CSTL/Marlin_2.1.x/files/9951239/MKS.Robin.Nano.V1.3_002.SCH.pdf)

riccalisci, 2022-11-23:
i tried to set at half SPI speed, but nothing has changed.

Foxies-CSTL, 2022-11-25:
Sorry but I can't help you more.

closed 2022-11-25 by Foxies-CSTL