Foxies-CSTL/Marlin_2.1.x #5: General question: Mesh and Probe XY Offset

ManticoreSigma, 2022-08-29:
Hi Foxies!
I know this is not an issue per se, so feel free to just close if annoying ;)
I've built an effector for the Q5 where the Z probe isn't dead centre below the nozzle. With your Marlin, I was able to input the XY offset, but I have no idea if the printer uses this offset for probing purposes.
Since I switched, it seems I have some first layer problems where there are areas on the bed that are too close to the nozzle, even though I ran the calibration wizards several times.
So... **TL;DR:** Does the firmware use the probe offset to adjust the probing points during the wizard?

Cheers,
Manti

Foxies-CSTL, 2022-08-31:
Hi @ManticoreSigma ,
Probe offset on Delta is still a problem because it introduces more or less parallax error depending on the probing location on the bed. The offset is taken into account during Delta calibration and also during leveling. To reduce this error, it is necessary to get closer to the center and be within 20mm of the nozzle tip. It is also possible to tilt the bed by performing a G29 J (Quick 3-point level) before printing, but this requires deploy/stow the probe each time.

ManticoreSigma, 2022-09-05:
Hey Foxies!
Thanks for clarifying. I'll try it out and see how much it affects my prints. Are the results interesting for you or is this such a niche issue that it's irrelevant?
Cheers,
Manti

ManticoreSigma, 2022-09-07:
BTW, as far as I can see, the G29 L1 does not load the correct UBL mesh. I used M420 L1 instead and that seemed to work.

Foxies-CSTL, 2022-09-08:
Yes this can also be done with the M420 but as I use Prusa/SuperSlicer, I use the G29Lx on my GCode filament and I did not notice your problem.
Maybe the save after polling (special Delta menu) was not done correctly.

ManticoreSigma, 2022-09-09:
It's really weird. Sometimes, the M420 L1 works, other time it doesn't, but G29 L1 is the way to go.
I reset the EEPROM yesterday and recalibrated everything. As soon as I reassemble the effector with the newer base plate, I'll try again.
Loading the mesh manually from the printer menus seems to work, though.

ManticoreSigma, 2022-09-11:
Okay, either I'm a complete idiot or there's a bug:
I did a UBL for both PLA and just default and manually saved the settings with M500. Then I let Octoprint output the Mesh data with M420 L1 and took a screenshot. The test print had a perfect first layer.
Then I restarted the printer and output the mesh data again. Same as in the screenshot.
But the first layer was again WAY too close to the bed. I tried the same GCode again, but with M420 L1 changed to G29 L1 manually in the file. Same result. I even entered M420 L1 and G29 L1 both manually before the print. Same again.
Then I did a Z calibration and the bed distance was still off. But this time, where the nozzle was too close before, it printed fine and where it printed fine before, it was too far away from the bed.

Is it possible that there's a bug that either doesn't load the Z offset or doesn't calculate the Z offset against the bed mesh when loading?
So either way, I can't get the mesh to load after restarting the printer. Every time I want to print, I need to recalibrate both Z and the mesh.
How do you normally load the mesh? I use the filament section in PrusaSlicer 2.5 to add the pre-GCode "G29 L1", nothing more.
The firmware version I use is Q5_8+SCWTPULR-Robin_nano.bin 2.1.1

Foxies-CSTL, 2022-09-13:
> I did a UBL for both PLA and just default and manually saved the settings with M500.

When you make a UBL mesh with an M500 then the mesh will default to the "0" location.

> I tried the same GCode again, but with M420 L1 changed to G29 L1 manually in the file. Same result. I even entered M420 L1 and G29 L1 both manually before the print. Same again.

So to choose the location of your mesh you have to save it afterwards with the command G29S "x" and recall/load it with the command G29L "x".

> s it possible that there's a bug that either doesn't load the Z offset or doesn't calculate the Z offset against the bed mesh when loading?

After adjusting a too large Z_OffSet then it is better to restart a mesh.

> So either way, I can't get the mesh to load after restarting the printer. Every time I want to print, I need to recalibrate both Z and the mesh.

Using the BabyStepping function to adjust the ZOffSet is not memorized and you must do it during or after the end of a printout because when you turn on your printer, it returns to the last memorized value.

closed 2022-11-25 by Foxies-CSTL