Foxies-CSTL/Marlin_2.1.x #6: [BUG] M118 Output to SERIAL_PORT_2

mulcmu, 2022-08-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

https://github.com/MarlinFirmware/Marlin/issues/24671

### Bug Timeline

https://github.com/MarlinFirmware/Marlin/issues/24671

### Expected behavior

https://github.com/MarlinFirmware/Marlin/issues/24671

### Actual behavior

https://github.com/MarlinFirmware/Marlin/issues/24671

### Steps to Reproduce

https://github.com/MarlinFirmware/Marlin/issues/24671

### Version of Marlin Firmware

https://github.com/MarlinFirmware/Marlin/issues/24671

### Printer model

https://github.com/MarlinFirmware/Marlin/issues/24671

### Electronics

https://github.com/MarlinFirmware/Marlin/issues/24671

### Add-ons

https://github.com/MarlinFirmware/Marlin/issues/24671

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

https://github.com/MarlinFirmware/Marlin/issues/24671

mulcmu, 2022-08-31:
See https://github.com/MarlinFirmware/Marlin/issues/24671 for additional discussion.

Foxies-CSTL, 2022-09-01:
I configured my printer with 4x2209Standalone + Wifi module with ESP2.1 + firmware BugFix2.1.1 => RAS.
I will do some tests and start working with the Wifi module and I will download your ESP firmware to do some tests.
See you later ;-)

mulcmu, 2022-09-01:
I create a new [branch ](https://github.com/mulcmu/Marlin_2.1.x/tree/mulcmu_v5)on my fork from your latest FLSUN-STABLE branch and made the configuration changes for my machine.  The bug behavior was confirmed when M118.cpp uses SERIAL_ECHOLN();

My ESPHome module would be a lot of work to setup unless you already have home assistant available to build it and test.  I can flash ESP3D 2.1 to my wifi module to help narrow down the root cause.

Foxies-CSTL, 2022-09-06:
> I create a new [branch ](https://github.com/mulcmu/Marlin_2.1.x/tree/mulcmu_v5)on my fork

Ok, I worked a little on the firmwares of the WIFI module. I tested the MKSwifi, Mischianti, ESP3Dv2 and V3 to define the properties of each in moniroting and file transfer. This allowed me to adjust the MoBo pins file according to the UI used (Marlin COLOR and MKS lgvl).
Finally I made tests between console/terminal and the tests are conclusive:
OCTO(M118 P2 Test P2 From OCTO) to ESP3D and I get well the "Test P2 From OCTO" in echo and vice versa. So it came from my pin definition and not from the M118.cpp file.
ESP3Dv3
[image]
OCTO
[image]
ESPD3v3
[image]

You can find all the modifications on my development branch "FLSUN_BUGFIX-MULTI".

closed 2022-09-08 by Foxies-CSTL