Foxies-CSTL/Marlin_2.1.x #8: [BUG] "Change Filament" command moves the carriages in strange ways

ManticoreSigma, 2022-09-07:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Used preconfigured firmware: Q5_8+SCWTPULR-Robin_nano.bin 2.1.1
When I select "Change Filament", the carriage in the back of my Q5 pulls upwards and the others drop slightly, but they are all under extreme stress. It also seems the printer wants to home, but moves one axis in the wrong direction, waiting for it to arrive, but it never does (the carriage collides with the top cover but keeps pulling upwards).
When selecting "Unload Filament" and then "Load Filament", everything works perfectly.

### Bug Timeline

First witnessed yesterday (September 6th 2022)

### Expected behavior

I expected the carriages to either stand still, home or just drop slightly.

### Actual behavior

The carriage on the "back axis" pulls upwards and the others drop different heights.

### Steps to Reproduce

Select the option "change filament"

### Version of Marlin Firmware

2.1.1

### Printer model

FLsun Q5

### Electronics

Stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Foxies-CSTL, 2022-09-08:
When there is an automatic or manual filament change, the head will be positioned at the back at "YMax - 20mm" to avoid purging the filament in the center of the bed. But this will change in the next versions and will be like this "M600 X0 Y-95 Z50" to take into account the bed size of the Q5

ManticoreSigma, 2022-09-09:
Okay, great! Then I'll wait for the next version and close the issue.

closed 2022-09-09 by ManticoreSigma