Foxies-CSTL/Marlin_2.1.x #11: [BUG] resets during G3 arc command

riccalisci, 2022-11-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After a new UBL mesh, untill now it has work, my FLSUN Q5 lock and resets during G3 arc (used at print start for wipe nozzle). 
Consideration:
1-If don't print G3 arc works. 
2-if deactivate UBL from display works

https://youtube.com/shorts/uBgh8Aw08mc?feature=share

### Version of Marlin Firmware

2.1.1

### Printer model

Flsun Q5

### Electronics

mks robin nano v1.3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

IdeaMaker

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration.zip](https://github.com/Foxies-CSTL/Marlin_2.1.x/files/10089843/configuration.zip)

Foxies-CSTL, 2022-11-25:
The ARC function is active on all my firmwares and must work with the G3. A tip: check if there is an error in your GCode Start purge line. 
Here is the purge line part I use with Super/PrusaSlicer:
`G92 E0.0; reset extrusion distance`
`G1 X-54.672 Y-95.203 Z0.3 F4000; go outside print area`
`G1 E2 F1000 ; de-retract and push ooze`
`G3 X38.904 Y-102.668 I54.672 J95.105 E20.999 ; fat 20mm intro line @ 0.30`
`G3 X54.671 Y-95.203 I-38.815 J102.373 E5.45800 ; thin +40mm intro line @ 0.08`
`G1 E-0.8 F3000 ;  retract to avoid stringing`
`G1 X52.931 Y-96.185 E0 F1000; {machine_max_feedrate_e[0]*6} ; -1mm intro line @ 0.00 ; wipe`
`G1 E0.6 F1500; de-retract`
`G92 E0.0 ; reset extrusion distance`

Otherwise I regularly use ArcWelder for all my circular objects and I haven't encountered your problem. 
Here are my settings:
Slicer => ArcWelder.exe --resolution-mm=0.05 --g90-influences-extruder --mm-per-arc-segment=1 --min-arc-segments=14 --default-xyz-precision=4 --allow-dynamic-precision --allow-3d-arcs --allow-travel-arcs
Octoprint => 
[image]
[image]

riccalisci, 2022-11-28:
my start G3 gcode is : "G3 X0 Y-100 I100 Z0.4 E15 F500". Ideamake is not able to generate arc gcode.		
In any case, i have reflash a new firmware and seem to works. I'll try to do a newone it based on yours start gcode.
let you know, thanks a lot.

Foxies-CSTL, 2022-12-04:
> Ideamake is not able to generate arc gcode.

Be aware that no current slicer can generate Arc gcode except with the ArcWelder plugin after slicing your model.

> let you know, thanks a lot.

A feedback of always appreciated, happy printing.

closed 2022-12-04 by Foxies-CSTL