Foxies-CSTL/Marlin_2.1.x #12: [BUG] Extruder stall on TMC2208_STANDALONE

jhstatewide, 2022-12-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Configuration: FLSUN QQS-PRO with TMC2208 on all axes. Configured standalone, dropped in as direct A4988 replacement.
Was running 2.x build of Foxies Marlin for quite some time, no issues. 
Read that LINEAR_ADVANCE was now supported for TMC2208 so decided to give it a try.

Switched mode to ALL_TMC8 and tried latest FLSUN-STABLE (commit ad8b84a9c8b1962f90ca7c609f53d16870950cb8)

After some delta calibration and setting of Z-offset, attempted print over USB. Seems like some stuttering on plain bottom layers, which seems funky. Print proceeds for 5 - 10 minutes and then a 'bang' is heard from the extruder and it halts for the rest of the print. The XYZ axes all continue to operate as normal.

Suspecting LIN_ADVANCE is not truly fixed, disabled it, recompiled, same behavior.

Reverted to last 2.1.1 release tag 29c5925d2c1cc32e98fb1ddd29853b40a1c1691e to see if behavior is any different. 
Behavior is the same, print starts semi-normally and then extruder halts shortly thereafter.

Reverted back to build with CONFIGURATION_H_VERSION 020008 (one of the Foxies Marlin 2.0.8 builds) and printer is normal.

I put many spools through the printer with TMC2208 on extruder on 2.0.8, no luck so far on 2.1.x with LIN_ADVANCE enabled or disabled.

### Bug Timeline

After upgrading to 2.1.x

### Expected behavior

Expected extruder to operate as normally

### Actual behavior

Extruder halted after 5 - 10 minutes of print

### Steps to Reproduce

Have FLSUN QQS
Install TMC2208 on all axes.
Install Marlin 2.1 on printer.
[config.zip](https://github.com/Foxies-CSTL/Marlin_2.1.x/files/10173088/config.zip)

### Version of Marlin Firmware

FLSUNQ_2.1.1 2022-08-06

### Printer model

FLSUN QQS-PRO

### Electronics

stock board with TMC2208 on all axes

### Add-ons

TMC2208

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Foxies-CSTL, 2022-12-07:
Hi @jhstatewide ,
I have looked at your attached files and I am asking for some clarification:
- Are you using MeatPack with Octoprint?
- Why did you activate it on the Wifi port (Rx/Tx = port2) and on the USB port (port1)?
=> MeatPack may be the source of your problems (Rx/Tx).
- Linear advance seems to be disabled so your problem with the TMC2208 is located elsewhere.
The rest of the settings seem to be correct except the SERIAL_2 port activated a second time with a different baudrate.
Try disabling MeatPack on port 2 and disable your "SERIAL_PORT_2 1" and "BAUDRATE_2 250K" lines in the config.h file.

jhstatewide, 2022-12-07:
> Are you using MeatPack with Octoprint?

I am using it with Octoprint.

Re: serial ports, I had some troubles getting the device to connect and went back and looked at my config.h from 2.0.8. I don't use the WiFi functionality at all (print exclusively through octoprint) so it's possible that I was just having a temporary serial issue and this setting affected nothing at all.

I ran this last config on 2.0.8 and just brought forth my changes. I can do a more "stock" build.
I am going to pull the latest 2.1.1 from this repo and try again.

jhstatewide, 2022-12-07:
@Foxies-CSTL PS it looks like someone upstream is having the same type of trouble as me. You can see it at the linked issue to the main Marlin repo.

jhstatewide, 2022-12-07:
Reflashed with ad8b84a9c8b1962f90ca7c609f53d16870950cb8

I see the same "stuttering" on plain old bottom layers, and then extruder halt a few minutes in. The last extrude before it halts is really loud, like a CLUNK. I'm not too sure what's going on because this hardware is fine with your older build based on 2.0.8.

I've attached the config zip and the diff below.

[config-211.zip](https://github.com/Foxies-CSTL/Marlin_2.1.x/files/10177714/config-211.zip)
Config here

diff:
```
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index ffda9462be..c20ed2e406 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -200,7 +200,7 @@
 
 // Name displayed in the LCD "Ready" message and Info menu
 #ifdef QQSP
-  #define CUSTOM_MACHINE_NAME "Delta QQSP"
+  #define CUSTOM_MACHINE_NAME "SAM Printer"
 #endif
 #ifdef Q5
   #define CUSTOM_MACHINE_NAME "Delta Q5"
diff --git a/Marlin/FLSUNQ_Config.h b/Marlin/FLSUNQ_Config.h
index 6dead4d3bc..8710995db9 100644
--- a/Marlin/FLSUNQ_Config.h
+++ b/Marlin/FLSUNQ_Config.h
@@ -45,12 +45,12 @@
 /*________________________2___________________________*/
           /*-----Type Stepper Drivers-(1 CHOICE)-----*/
 /* MODE STOCK for QQS & Q5 & SR */
-#define STOCK                         // (S) (Default_QQS) For 4xA4988(green or red color).
+// #define STOCK                         // (S) (Default_QQS) For 4xA4988(green or red color).
                                       // (S) (Default_Q5) For 3xTMC2208+1xA4988.
                                       // (S) (Default_SR) For 4xTMC2209_UART.
 
 /* MODE STANDALONE XYZ+E for QQS & Q5 */
-//#define ALL_TMC8                    // (8) For 4xTMC2208_STANDALONE
+#define ALL_TMC8                    // (8) For 4xTMC2208_STANDALONE
 //#define ALL_TMC9                    // (9) For 4xTMC2209_STANDALONE
 
 /* MODE UART XYZ+E for QQS & Q5 */
@@ -100,7 +100,7 @@
 * =========================================//
 */
 //#define LCD_LANGUAGE en                // Change for your country ('bg':'Bulgarian', 'ca':'Catalan', 'cz':'Czech', 'da':'Danish', 'el':'Greek', 'fi':'Finnish', 'hr':'Croatian', 'hu':'Hungarian', 'jp_kana':'Japanese', 'nl':'Dutch', 'pl':'Polish', 'pt_br':'Portuguese (Brazilian)', 'ro':'Romanian', 'ru':'Russian', 'sk':'Slovak', 'sv':'Swedish', 'tr':'Turkish', 'uk':'Ukrainian', 'vi':'Vietnamese', 'zh_CN':'Chinese (Simplified)', etc)
-//#define BOOT_MARLIN_LOGO_SMALL         // Small Logo Marlin to reduce de binary. Comment to have normal LOGO(Default).
+#define BOOT_MARLIN_LOGO_SMALL         // Small Logo Marlin to reduce de binary. Comment to have normal LOGO(Default).
 
 /*_________________________________4______________________________*/
           /*---- Extruder, Custom effector and Modules -----*/
@@ -190,7 +190,7 @@
 
 /*_______________________6____________________*/
   //======Many options for Modules: ========//
-#define LIN_ADVANCE                      // (L) (Default2209) with K=0 For TMC_UART2208 prefer mode spreadCycle(by TFT menu) or commented if problem.
+// #define LIN_ADVANCE                      // (L) (Default2209) with K=0 For TMC_UART2208 prefer mode spreadCycle(by TFT menu) or commented if problem.
 #define POWER_LOSS_RECOVERY              // (Default) Continue print after Power-Loss.
 
 //=================================================================================//
@@ -211,14 +211,14 @@
 * == Tip: commented the line "BOOT_MARLIN_LOGO_SMALL" for more space EEPROM ==
 * ============================================================================
 */
-//#define OCTO                           // Enable buffer for Octoprint.
+#define OCTO                           // Enable buffer for Octoprint.
 #define HOST_ACTION_COMMANDS             // Default - Action Command Prompt support Message on Octoprint
 #define HOST_START_MENU_ITEM             // Add a menu item that tells the host to start a print
 
-#define BINARY_FILE_TRANSFER             // Bin transfert for ESP3D firmware v2.1 or others.
+// #define BINARY_FILE_TRANSFER             // Bin transfert for ESP3D firmware v2.1 or others.
                                          // Not compatible with the MEATPACK option. 
 //------ Support for MeatPack G-code compression (OCTOPRINT)--------//
-//#define MEATPACK_ON_SERIAL_PORT_1      // (M) With connection USB
+#define MEATPACK_ON_SERIAL_PORT_1      // (M) With connection USB
 //#define MEATPACK_ON_SERIAL_PORT_2      // With other connection like Tx/Rx Wifi socket.
 
 //----------Options Plus-----------//
```

Foxies-CSTL, 2022-12-08:
> I see the same "stuttering" on plain old bottom layers, and then extruder halt a few minutes in.

Can you provide me with the M503 output file when you see the issue?

> I've attached the config zip and the diff below.

No remark except the disabling of LA. As you don't have a Wifi module, you can also disable/comment the line "#define MOD_WIFI "

jhstatewide, 2022-12-09:
I'll keep you posted if I give this a shot. I think I'm just going to ditch the TMC2208 in favor of TMC2209, which seem to be less problematic. Thanks for your attention. There is some minor activity upstream that makes me suspect it is an upstream issue and nothing to do with your delightful fork.

closed 2023-01-21 by Foxies-CSTL

jhstatewide, 2024-02-18:
@Foxies-CSTL Necroing this from a long time ago. Put TMC2209 on the extruder and it's been perfect since then.