Foxies-CSTL/Marlin_2.1.x #13: Printer stuck in initialising loop if SD card is inserter during power on.

Blastaar-BR, 2023-01-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi noticed that with this new 2.1.1 version, if we turn on the QQSPro printer with the SD card still on the slot, the printer try to initialise, then complete but restart again and again. If I take out the SD card from the slot and cycle the power, it works. [BUG] (bug summary)

### Bug Timeline

New bug

### Expected behavior

When power up even with SD card, the printer should initizalise normally, without need to take off the SD card from the slot.

### Actual behavior

Get stuck on a initialising loop, restating again and again.

### Steps to Reproduce

1 - With printer off, introduce any SD card with Gcodes.
2 - Turn printer on
3 - The printer screen shows Marlin draw , initialising, complete then restart again and again.
4 - Taking off the SD card and cycling the power, the printer get back to the normal.

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

FLsun QQS pro

### Electronics

Stock electronics

### Add-ons

No

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Foxies-CSTL, 2023-01-21:
Did you try with a SD card of less than 16GB formatted in FAT32 block 4096 and Class10 (fast read/write)? 
Note: Do not use the SD card provided (Class4) because it is not of good quality for printing only to keep the doc and files.

peeepeeepeee, 2023-01-29:
just flashed the same version and had the same problem - cause by ROBIN_MINI.CUR on the SD-card

Foxies-CSTL, 2023-02-04:
> just flashed the same version and had the same problem - cause by ROBIN_MINI.CUR on the SD-card

After the flash and remove the SD card, you should have the screen calibration. After that, I advise you to reset the EEPROM via the "Special Delta" menu. There will be a reboot and a new screen calibration request but the eeprom will be cleaned.

closed 2023-04-09 by Foxies-CSTL