Foxies-CSTL/Marlin_2.1.x #14: Can not find my firmware version anymore

pingywon, 2023-01-20:
### Is your feature request related to a problem? Please describe.

I can not find the version of firmware  that I need for my specific machine set up.

I have used "SCWBPULR-Robin_mini.bin" and it is no longer included in the package? Am I supposed to use something else? If so, which one?  Thank you

### Describe the feature you want

need variant SCWBPULR-Robin_mini.bin for my machine.

### Additional context

Please don't tell me it needs to be build from source. :/

pingywon, 2023-01-20:
It looks like I can find my version but with the "W". Im not sure about this flag. Can I assume that that is the right version I need? Thanks!

pingywon, 2023-01-20:
Yeah my bad. I must have wrote it down wrong back last time I figured out my firmware.

Foxies-CSTL, 2023-01-21:
This firmware "SCWBPULR-Robin_mini.bin" is for QQSP Stock with reverse direction and it exist on 2.1.1 
https://github.com/Foxies-CSTL/Marlin_2.1.x/blob/Firmwares/QQSP/SCWBPULR-Robin_mini.bin
https://github.com/Foxies-CSTL/Marlin_2.1.x/raw/Firmwares/QQSP/SCWBPULR-Robin_mini.bin

closed 2023-01-21 by Foxies-CSTL