Foxies-CSTL/Marlin_2.1.x #18: Stock QQSP Goes Crazy on Foxies!

beebenutz, 2023-03-12:
After successfully completing the bed leveling procedure, I begin a new print.
As the print begins, I attempt to adjust the Nozzle to probe offset to correct the first layer height.
If The offset is adjusted, the printer begins crashing into the bed and lifting off on random trajectories. 

This can be replicated by flashing the firmware (2.1.2) (QQSP_STOCK-Robin_mini)
Conduct a bed leveling procedure. (I used UBL)
Then begin a print and attempt to adjust the nozzle to probe offset.

Foxies-CSTL, 2023-04-06:
I can help you but I need some details:
- Did you change the stepper motor drivers?
- After flashing the firmware, did you force the erasing of the EEPROM by the menu "Special Delta" (first item)?
- To adjust the nozzle height during the first printing layers, you have to use the "BabyStep" function (3 ways to do it => note on the Wiki)

mlewis-everley, 2024-01-04:
I have also been getting this issue on my Q5's. It seems like one of the steppers (possibly Z) locks in place (or behaves erratically) when using "tune" > "z offset" options in the LCD menu.

mlewis-everley, 2024-01-05:
OK, so I have updated to the latest stable version. Now, instead of random movements, the printers just turn off their steppers (so the effector stops moving and I can freely move it by hand) :-/.

Foxies-CSTL, 2024-01-16:
Hi,

> 3\. I cannot find anything related to "Babystepping"  in the wiki. I usually just use "Tune" > "Probe Z Offset" in the LCD menu while printing.

You'll find it here:  https://github.com/Foxies-CSTL/Marlin_2.1.x/wiki/2.SETTINGS-THE-PRINTER#222-tips-babystepping

[image]

> OK, so I have updated to the latest stable version. Now, instead of random movements, the printers just turn off their steppers (so the effector stops moving and I can freely move it by hand) :-/.

You may have a Vref setting problem on the stepper drivers. If they are removable, you have a micro-screw to adjust, otherwise it's another problem.

mlewis-everley, 2024-01-16:
Thanks for the reply. I am using removable drivers with the microscrews. I cannot remember off the top of my head what voltage it is set to, what do you recommend @Foxies-CSTL ?

Foxies-CSTL, 2024-01-16:
> Thanks for the reply. I am using removable drivers with the microscrews. I cannot remember off the top of my head what voltage it is set to, what do you recommend @Foxies-CSTL ?

For A4988s, Vref should be between 0.65 and 0.79v.
For TMC220xs, Vref must be between 0.8 and 1v.

Mine are 960mV for the XYZE axes, but I set them in the UART by current (Irms) and currently I have 900mA on XYZ and 850mA on Ext.

closed 2024-03-14 by Foxies-CSTL