Foxies-CSTL/Marlin_2.1.x #20: [BUG] Error compiling firmware for mks_robin_nano_v1_3_f4

Doct0rDuc, 2023-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

PLATFORM.IO don't build the firmware for my FLSUN Q5(2021-x):
-mobo mks robin nano v1.3(stock)
-axis driver stock, extruder driver TMC2208V3.

I've correctly changed the motherboard in the Configuration.h file to BOARD_MKS_ROBIN_NANO_V1_3_F4 and i've changed the enviroment of platformio in the platformio.ini file to mks_robin_nano_v1_3_f4.
I've only edited FLSUN_config.h and Configuration.h files.

Platformio.ini:
[image]

Configuration.h motherboard define:
[image]

if i try to build the firmware using the older motherboard(and changing env in platformio.ini) the firmware is build successfully
[image]

### Bug Timeline

new one, it's started with the sourcecode version 1.3.0

### Expected behavior

i expected to compile my firmware with no problems

### Actual behavior

vscode terminal print an runtime error during firmware building

### Steps to Reproduce

-changed env in the platformio.ini file to mks_robin_nano_v1_3_f4
-edited FLSUNQ_Config.h for my printer selecting decommenting "#define Q5" and commenting "#define QQSP"
-changed the mobo version to "BOARD_MKS_ROBIN_NANO_V1_3_F4"  in Configuration.h
-start building the firmare
-Runtime error stop the build:

```
RuntimeError: deque mutated during iteration:
  File "C:\Users\feder\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\feder\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\feder\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\Users\feder\Desktop\Marlin_2.1.x-1.3.0\buildroot\share\PlatformIO\scripts\offset_and_rename.py", line 28:
    marlin.replace_define('STM32_FLASH_SIZE', maximum_flash_size)
  File "D:\Users\feder\Desktop\Marlin_2.1.x-1.3.0\buildroot\share\PlatformIO\scripts\marlin.py", line 19:
    for define in env['CPPDEFINES']:
```

### Version of Marlin Firmware

2.1.x

### Printer model

Flsun q5 2021-

### Electronics

stock board, changed the exdruder and the Y driver to 2 TMC2208 v3

### Add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/Foxies-CSTL/Marlin_2.1.x/files/11440962/Config.zip)
firmware changes:
-all 4driver setted to 4TMC
-disabled Wifi module
-diabled host setting
-disabled "DELTA_HOME_TO_SAFE_ZONE" (sometimes when i print high things the hotend go down and touch it)
-setted "#define TEMP_TIMER  5"  as the comment for Q5(2021-) says
-enabled the display standby and uncommented "#define TOUCH_IDLE_SLEEP_MINS 5"

riccalisci, 2023-05-11:
Help, same problem form me. until last week it works perfectly

Doct0rDuc, 2023-05-12:
after a day spent looking for a solution on the internet, in the end I went to the official marlin github and downloaded the blank sourcecode, edited it's two configuration files using the ones already made by foxies plus the changes I needed and after hours scrolling the code I finally builded the firmware I wanted.
But for the problem of not compling the foxies code with my motherboard i did't found a solution

closed 2023-05-12 by Doct0rDuc

riccalisci, 2023-05-13:
Hi Doc0orDuc, could be upload on your GitHub the source code that you are created?  it would help me to solve the problem.
Thanks a lot

riccalisci, 2023-05-13:
tried to begin from original source code and after a few hours, same identical problem :(

Doct0rDuc, 2023-05-13:
> Hi Doc0orDuc, could be upload on your GitHub the source code that you are created? it would help me to solve the problem. Thanks a lot

I've uploaded the source code on github

riccalisci, 2023-05-13:
Thanks a lot!!