Foxies-CSTL/Marlin_2.1.x #21: [BUG] FILAMENT SENSOR AND POWERLOSS RECOVERY NO WORK

matiasdj87, 2023-05-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

FILAMENT SENSOR NOT WORK and POWERLOSS RECOVERY NOT WORK.... with SCWTPULR-Robin_mini.bin STOCK FIRMWARE... but with QQS Repetier Firmware its works fine. M412 is ON and M119 is tiggered or open if the filament is charged or no..

### Version of Marlin Firmware

2.1.2

### Printer model

FLSUN QQS PRO

### Electronics

MKS ROBIN MINI

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Foxies-CSTL, 2023-06-25:
Hi,
These two options are available via the "Configuation" menu ("Runout" and "power_Outage") and you need to activate them (toogle button) and then save their position by saving settings (configuation menu).

cpeg78, 2023-08-03:
Hello, when I turn on runout, the printer is stopping and wants to change filament. Filament was correctly loaded and sensor light is blue.

Foxies-CSTL, 2023-08-30:
> when I turn on runout, the printer is stopping and wants to change filament. Filament was correctly loaded and sensor light is blue.

Hi,
You can check the correct operation of the filament sensor with the M119 command, which will give you an instantiation of the ON/OFF status in the terminal.

[image]

If you don't have the "TRIGGERED" state when the filament is inserted, then you've got reverse detection.

cpeg78, 2023-08-30:
Hi,thanks for your answer. Is it possible to change the detection? On my Ender 3 V2, I can change it from High to Low, but can‘t find such setting on my Q5.

Foxies-CSTL, 2023-08-30:
> Is it possible to change the detection?

Yes, this is possible in the firmware, but it may be easier to invert the wires if you have a 2-wire detector: 1 wire for signal and 1 wire for ground.

Foxies-CSTL, 2023-08-30:
> I can change it from High to Low, but can‘t find such setting on my Q5

Yes here in the "Configuration.h" file

[image]

cpeg78, 2023-08-30:
I already had this idea. Its a cheap on from ebay, with 3 wires.

Foxies-CSTL, 2023-08-30:
> I already had this idea. Its a cheap on from ebay, with 3 wires.

I answered earlier ;-)

closed 2024-03-14 by Foxies-CSTL