Foxies-CSTL/Marlin_2.1.x #23: 2.1.2 Screen Calibration Issue

J14111, 2023-10-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

Thanks for the great work, I have just upgraded to the latest version, I have a QQS-PRO using the TM9 file it flashes fine.

I am prompted to calibrate the screen which I do, top left, bottom left, top right, bottom right but after this it goes back to top left but there is also a cross in the bottom right and I cant press anywhere it gets stuck on this screen?

### Bug Timeline

new 2.1 was fine

### Expected behavior

expect to get past LCD calibration

### Actual behavior

Stuck on LCD calibration

### Steps to Reproduce

1. Flash firmware
2. Try to calibrate screen

### Version of Marlin Firmware

2.1.2

### Printer model

QQS Pro

### Electronics

stock

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

none

Foxies-CSTL, 2023-10-13:
Hi @J14111  ,
Calibrating the screen isn't easy with your fingers ;-) For my part, I use a stylus or pen cap to be precise, tapping the center of the cross. If it doesn't work the first time, switch off your printer and try again.

Jamesp141, 2023-10-13:
Thanks i will try that next time, in the end i built my own from your files and marlin, it worked first time