Foxies-CSTL/Marlin_2.1.x #25: Files not printing from SD

biged83, 2024-03-04:
I just imaged my FLSUN QQS pro with the 2.1.2 firmware. When i try to access the SD card i have to hit refresh multiple times then when the files load i can select them but nothing prints. It doesnt start the print. I have tried using cura and pursa slicers.

biged83, 2024-03-08:
Still having this issue. Even using the tmc2208 firmware.

Foxies-CSTL, 2024-03-10:
Hi,
Do you start to heat up when you select a file to print?
Make sure you use short file names (24characters) and check that your starting GCodes don't set the extruder to absolute mode (M82 vs M83).

biged83, 2024-03-10:
It never heats up. But I think my mainboard was bad. Replaced it and got it all working on stock firmware. May try upgrading firmware once I get a new level

Foxies-CSTL, 2024-03-10:
Have you tried reflashing the original firmware before replacing your MoBo? 
Maybe it was the new MoBo HiSpeed with the GD32 chipset instead of the STM32.