Foxies-CSTL/Marlin_2.1.x #26: Q5_8+SCWXPULRM-Robin_nano.bin  - no direct drive (extruder to fast)

chromoxdor, 2024-03-13:
My Q5 got a direct drive extruder (micro swiss) so i decided to use the above mentioned firmware file to update from the stock firmware. Unfortunately the speed when loading filament is way too high. It looks like this is still set for a geared extruder. Or am i doing something wrong and there is a  setting i missed?

Foxies-CSTL, 2024-03-14:
Hi,

your extruder has a different gear ratio than the original (1:1 vs 3:1) so you need to set the Estep steps to 130 instead of 410.

You can do this in the "Steps/mm" section of the "Configuration/Advanced settings" menu. Then save it, but don't forget that this is only a starting point.  The exact value can be determined by performing an ESteps setting test, of which here's a link.
https://github.com/Foxies-CSTL/Marlin_2.1.x/wiki/2.SETTINGS-THE-PRINTER#26-extruder
or
https://teachingtechyt.github.io/calibration.html#esteps

chromoxdor, 2024-03-14:
Thanks... this is working. 
But after unloading filament the head is always crashing into the bed. any ideas why this is happening? 
I also tried it with the stock settings to see if it isn´t an issue with changing the e-steps. Same problem..

chromoxdor, 2024-03-14:
BTW: Loading filament works fine

Foxies-CSTL, 2024-03-14:
If your Z_Offset setting is ok, then you may have forgotten to load the correct mesh into your GCode Start (G29 L1 = for PLA mesh).
Here's a reminder.
https://github.com/Foxies-CSTL/Marlin_2.1.x/blob/Firmwares/Instructions.md

chromoxdor, 2024-03-14:
No, this seems to be kind of a bug.
To reproduce: Turn on the printer and without doing anything else unload the filament.
Usually the printer lifts the head a bit when unloading. This doesn´t happen when freshly turned on!
But the printer does want to move the head down again... 😳

chromoxdor, 2024-03-14:
I also have a small side question. I successfully build my own firmware. It works so far, but i also want the background to be black. 🙂  Where do i define this?

Foxies-CSTL, 2024-03-14:
> But the printer does want to move the head down again... 😳

This is normal, as a Delta doesn't know where it is at start-up, unless you do a "Home" before operations;

> but i also want the background to be black. 🙂 Where do i define this?

The "Theme" option will be available with future 2.1.3 ;-) but if it's just the background, you'll need to insert a line in your build:
#define COLOR_BACKGROUND COLOR_DARK

Foxies-CSTL, 2024-03-14:
> * turn on printer -> start unloading filament -> unloading finished an head lowers and crashes
> 
> * turn on printer -> homing -> steppers off -> head moves to buildplate because of gravity ->  start unloading filament -> head moves up a bit -> filament unloads -> unloading finished head moves down a bit.

There are options to activate in the main configuration to block the Z axis, but there is also an adjustment to be made on your eccentric wheels to prevent the assembly from descending towards the bed.

Otherwise, filament change is automatic in two sequences, whatever the position of the head: a park movement (XYZ) then an E movement for loading/unloading the filament.

> > The "Theme" option will be available with future 2.1.3 ;-)
> 
> That sounds interesting :)

* ===========News feature for V2.1.3 ==========
* Change EEPROM reset with disable(M593 F0) IS to prepare your Delta without IS enable. 
* Add mode portrait for TFT32/TS35.
* New MoBo temp.
* New stm32f1.ini (hispeed)
* add state time/restant
* Show the E position (filament used) during printing
* Show the last file on first place.
* New theme DELTAFOX.
* disable UBL_HILBERT and enable UBL_TILT
* New menu for filament (runout lengh, time)
* New menu for Auto-Sleep screen to 0 at 99mn screenview. (M255 S5)
* etc

Foxies-CSTL, 2024-03-18:
> so i am not sure where to ask this other than here.

I'm on Discord on the Marlin or FLSUN server.
Flsun = "https://discord.gg/m8hbjpTM49"
Marlin = "https://discord.gg/SucfDN8Z"
See you soon ;-)